GyverLibs/ServoSmooth #1: функция setTargetDeg() занижает TARGER_DEGREE значение при движении от меньшего к большему углу

xillmera, 2021-07-23:
Эта проблема существует на примере — "../ServoSmooth-main/examples/SmoothSweepTest/SmoothSweepTest.ino" .
При движении __ PRESENT_ANGLE → TARGET_ANGLE : 50 → 10 __ движение происходит до правильного конечного угла REAL_TARGET_ANGLE = TARGET_ANGLE   с минимумом ошибок (как будто условием является больше не равно). Хотя само значение TARGET_ANGLE на 1 (единицу) меньшее. Такое возможно, к примеру, в случае использование условия for(;REAL_TARGET_ANGLE > TARGET_ANGLE; REAL_TARGET_ANGLE--).
Однако при движении __ 10 → 50__ во первых сам TARGET_ANGLE уменьшается на 1 (вместо 50 → 49), так и возникает часто проблемы доводки до этого значения как для четных, так и для нечетных значений углов. REAL_TARGET_ANGLE+1 = TARGET_ANGLE. При этом существуют моменты доводки REAL_TARGET_ANGLE до угла в 49 град, но они делят событийный ряд с частыми 48 град с периодичностью 1:1 (т.е 50% ошибок на каждый результат, если учитывать что 49 град. тоже ошибка).
Подобная проблема является критической для задач с зацикливанием платформы (ARDUINO PRO MINI ATmega328P 5V 16MHz) на цикле while(servo.getTargetDeg() == servo_angle_max) при доводке сервопривода(SG90) до необходимого угла.

AlexGyver, 2021-07-23:
Можно пример кода? Здесь какие то абстракции

xillmera, 2021-07-24:
@AlexGyver 
```
void loop() {
  Serial.print(servo.getCurrentDeg());
  Serial.print(' ');
  Serial.print(servo.getTargetDeg());
  Serial.print(' ');
  Serial.println(servo.tick() * 10);	// состояние серво третьим графиком
  delay(10);

  static uint32_t timer;
  if (millis() - timer > 3000) {
    static bool kek = false;
    kek = !kek;
    timer = millis();
    servo.setTargetDeg(kek ? 10 : 50);
  }
}
```
видос с выводом можно обнаружить по ссылке (если он, конечно, необходим)