GyverLibs/ServoSmooth #2: функция getCurrentDeg() возвращает не настоящее, а теоретическое значение

xillmera, 2021-07-23:
Если заблокировать движение рычага сервопривода на пути до TARGET_ANGLE не произойдет никаких изменений в показаниях getCurrentDeg() и функция вернет значение TARGET_ANGLE даже если сам сервопривод до него не смог довести рычаг. Это возможно в том случае, если изначально есть время выполнения операции и шкала времени дискретезируется по количеству доступных значений углов (без учета секунд и минут) (когда для соответствию углу выделяется область).

AlexGyver, 2021-07-23:
Эм, сервопривод не имеет обратной связи по углу))

xillmera, 2021-07-24:
@AlexGyver, эх жаль... Только вот одно не вяжется. Как он тогда в начале может на конкретный угол установиться при любом входном...

AlexGyver, 2021-07-24:
У него самого обратная связь есть внутри, но сама серво может только принимать сигналы