GyverLibs/ServoSmooth #3: Нельзя использовать аппаратный таймер по каналу А?

degorychev, 2021-07-27:
У меня основной цикл программы довольно долгий (чуть меньше секунды изза опроса 30ти датчиков ds18b20). В результате чего я вызываю servo0.tick() довольно редко и они двигаются очень медленно.
По этому я предположил что могу вызывать tick из аппаратного прерывания по таймеру (Можно ли в моем случае так делать?).
Но когда я в программе объявляю функцию ISR(TIMER5_A) {} (Из библиотеки GyverTimers) среда не компилирует код ссылаясь на ошибку 

> ...libraries\Servo\avr\Servo.cpp.o (symbol from plugin): In function `ServoCount':
(.text+0x0): multiple definition of `__vector_47'

Хотя при этом используя  ISR(TIMER5_B) код вроде скомпилировался, пока не загружал.
Можете подсказать чем вызвано такое поведение?

AlexGyver, 2021-07-27:
Ds18 можно и нужно опрашивать асинхронно. Смотри примеры к библиотеке, либо используй мою microds18b20, там есть пример с таймером. Также стоит вообще освоить конструкцию с программным таймером, 99.9% задач решается с его помощью.
По поводу аппаратных таймеров - какая плата? Мега?

degorychev, 2021-07-27:
> Ds18 можно и нужно опрашивать асинхронно. Смотри примеры к библиотеке, либо используй мою microds18b20, там есть пример с таймером. Также стоит вообще освоить конструкцию с программным таймером, 99.9% задач решается с его помощью.
Использую microds18b20, да. Асинхронных примеров незаметил если честно, эту часть кода писал месяца полтора назад, посмотрю... (она кстати странно себя вела когда на линии висело больше 15 датчиков, но сделал жгуты по 5 штук и повесил на разные пины)
> По поводу аппаратных таймеров - какая плата? Мега?
Да

AlexGyver, 2021-07-27:
Асинхронный я добавил, и правда не было. Придется тогда в него добавить счётчик, чтобы опрашивать по очереди, возможно чуть переделаю библиотеку чтобы это было удобнее.
На Меге серво использует пятый таймер, очень странно зачем :D

degorychev, 2021-07-27:
> На Меге серво использует пятый таймер, очень странно зачем :D
Да она не только с пятым не компилируется, а с любым (?)
Сделал опрос датчиков через милис, я даже не обратил внимания что там delay стоит, спасибо за наводку, посмотрим на сколько все станет хорошо.
Если причина такого поведения вам не ясна - задачу наверное можно закрывать.

PS: У вас кстати в readme этого репозитория опечатка 

> **Устанвока** целевой позиции серво по углу...

GyverLibs, 2021-07-27:
градусах в секунду в квадрате, вроде всё верно) я сейчас допилю микродс18 для удобного опроса грозди датчиков в цикле

degorychev, 2021-07-27:
> градусах в секунду в квадрате, вроде всё верно) 

может тогда уж градусов/сек^2?

>я сейчас допилю микродс18 для удобного опроса грозди датчиков в цикле

Было бы удобно, буду ждать. Заодно если есть возможность подключите 20 датчиков одновременно, у меня они не хотели читаться вашей библиотекой, хотя при этом адреса читались все...

GyverLibs, 2021-07-27:
> может тогда уж градусов/сек^2?
Да, но хозяин барин, сек*сек == сек^2 =)
У меня нет 20 датчиков к сожалению, тестировал 5 - всё работало

degorychev, 2021-07-27:
> У меня нет 20 датчиков к сожалению, тестировал 5 - всё работало

При 15 тоже все работает, а вот подключая жгут еще на 5, все ломается :D

GyverLibs, 2021-07-27:
ну ща починим, ты главное не уходи далеко)

GyverLibs, 2021-07-27:
вотс архив с тестовой версией, пример read async many. Датчики создаём массивом, адреса тоже, в итоге кода на пять строк там. Проверил на своих 5ти, всё работает, коменты в примере есть https://yadi.sk/d/2hT9samGQkTkkg

GyverLibs, 2021-07-27:
серво с таймером у меня кстати компилит, если выбран не таймер, который забирает серво

degorychev, 2021-07-27:
> вотс архив с тестовой версией, пример read async many. Датчики создаём массивом, адреса тоже, в итоге кода на пять строк там. Проверил на своих 5ти, всё работает, коменты в примере есть https://yadi.sk/d/2hT9samGQkTkkg

ну на данный момент у меня стенд собран уже из 4 кос по 5 датчиков в разные пины, и так наверное и останется (читается же заодно быстрее, а пинов у меги много). Но для теста попробую спаять переходник-объединитель на днях, заведу ишьюс в той библиотеке если найду проблемы. 
Спасибо! А вот в ридми в библиотеке SmoothServo все же исправьте, глаз так и цепляется за эту "устанвоку"

> серво с таймером у меня кстати компилит, если выбран не таймер, который забирает серво

Не компилит с любым таймером, когда объявляется любая функция ISR(TIMERx_A) {} ни первый ни пятый. 
Но с милис впринцепи это уже не проблема

GyverLibs, 2021-07-27:
> Не компилит с любым таймером

блин, у меня компилит))

degorychev, 2021-07-27:
> > Не компилит с любым таймером
> 
> блин, у меня компилит))

Чудно, ладно. попробую разработать минимальный код, с которым не компилит... (Проект большой, библиотек довольно много в итоге, может быть в чем то другом проблема)

GyverLibs, 2021-07-29:
не пробовал микродс запустить из архива?

closed 2024-01-25 by GyverLibs