GyverLibs/ServoSmooth #5: ServoSmooth.h - Не работает smoothStart

ZernovTechno, 2023-08-08:
Используется - ESP8266 nodeMCU с CP2102
Ядро для ESP8266 - установлено из IDE - 3.1.2
Версия IDE - 2.1.1

Данной функции в примерах я не увидел, может слепой :)

Код: 

```
#include <ServoSmooth.h>
ServoSmooth servo;

void setup() {
  servo.attach(0);      // подключить
  servo.setSpeed(10); // ограничить скорость
  servo.smoothStart(); //"Поехали!"
}

void loop() {
  servo.tick(); //тик-так
}
```

Серво летит в позицию моментально, а не с ограниченной скоростью. 
Проблема не у меня одного, тут уже встречался такой же несчастный:[ *тык*](https://community.alexgyver.ru/threads/voprosy-po-biblioteke-servosmooth.3364/page-2)
 
 Серво подключен на пин 0 (D3), серво и плата питаются от одного источника - 5в 2.4а, просадок по напряжению не было.
Провода качественные. Плавное движение в других частях кода работает (setTargetDeg())

GyverLibs, 2023-08-08:
возможно в esp либе другая реализация attach/detach. В классической attach сразу вертит серво на текущий угол (по умолч 0), smoothStart по сути дёргает attach detach с паузами, и серво приползает небольшими рывками к нулю, точнее к заданному setCurrentDeg

ZernovTechno, 2023-08-08:
Я, кажись, разобрался. Servo в ESP8266 (Лежит в папке библиотек ядра ESP8266) в attach вызывает самый обычный write на нужный угол. По дефолту там стоит "какой то", но есть перегруженная версия метода attach, которой можно передать угол на который крутить.  В общем вариантов много - можно  (все операции проводить с Servo.cpp внутри библиотек ядра ESP8266):

1. Если вам вообще не надо на attach крутить серво - удалите строку write(value); ( 84-ая в Servo.cpp, находится в методе Servo::attach(int pin, uint16_t minUs, uint16_t maxUs, int value) ).

2. Если нужно крутить плавно - можно заменить write(value) в Servo.cpp костылём из цикла for.

3. По идее можно как-то всунуть методы write из библиотеки гайвера в Servo.h.

А вообще, очень надеюсь что библия не заброшена и жду апдейта с решением этой задачи, который, очевидно, должен быть.

GyverLibs, 2023-08-08:
А detach корректно работает? SmoothStart работает исходя из того, что детач сразу бросает серву, поэтому если дергать аттач+детач - серва плавно поедет к углу, на который отправляет аттач