GyverLibs/ServoSmooth #6: Библиотека не позволяет сервоприводу вращаться в заданном диапазоне.

THEb0nny, 2023-12-04:
Есть сервопривод от geekservo, который вращается в диапазоне от 0 до 360 градусов.
[image]
Сам сервопривод корректно работает с стандартной библиотекой, которая с этим сервоприводом методом микросекондс, предварительно задав максимальное и минимальное значение диапазона импульсов.
https://www.youtube.com/watch?v=OYC-iuB75KA

Диапазон получается примерно такой, в котором он функционирует с этой библиотекой. Причём с одной стороны угол меньше, чем с другой стороны.
[image]

Вот наше видео, сервопривод движется не в максимальном диапазоне.
https://photos.app.goo.gl/wmh89MsZCgYEfZdi7
Вот наш код.
https://github.com/THEb0nny/Manipulator-palletizer/blob/main/sketch_apr13_manipulator-palletaizer/sketch_apr13_manipulator-palletaizer.ino

THEb0nny, 2023-12-07:
@AlexGyver можете посмотреть?

GyverLibs, 2023-12-07:
Ничего не понял если честно

GyverLibs, 2024-03-10:
увеличить диапазоны минимума и максимума

THEb0nny, 2024-03-11:
> увеличить диапазоны минимума и максимума

Увеличивал вроде как. И в характеристиках 500-2500 у них. Со стандартной библиотекой с этими диапазонами работает всё...

GyverLibs, 2024-03-11:
тогда не знаю, servosmooth крутит мотор при помощи стандартной servo либы. В следующей ревизии проверю на всех платформах и добавлю поддержку других серво-библиотек