GyverLibs/ServoSmooth #7: esp + servo

mcsham, 2023-12-30:
Привет всем.  С наступающим. 
у меня проблема использование этой библиотеки и esp8266.
На ардуинке, поворот работает как надо, указал укол и пошло поплыло, esp максимум на 90 градусов. 
Использую MG995 как серво и не понимаю почему такая проблема. 
Возможно проблема, что серво хочет 5в управления, а esp дает 3.3, или же это проблема с ШИМ.
конечно можно взять связку esp  + arduino но хотелось бы обойтись только esp т.к. цель одна, управлять дистанционно сервой. 
Что на esp 01s что на esp-13f, эффект одинаковый.
Схема тривиальная, но все же.

[image]

GyverLibs, 2024-01-25:
Стандартная Servo h работает корректно?

aleta2020, 2024-01-29:
Привет всем! Аналогичная проблема с TowerPro gs90 на китайском WEMOS D1 (ESP8266MOD): стандартная библиотека Servo.h работает (при min=450, max=2650) на 0-180 градусов - вращение быстрое. 
При использовании ServoSmooth (стандартный пример SmoothSweep на разных пинах, на 2х МК D1 и на 3x sg90) отклоняется +-45  (т.е. общий размах не более 90) и низкая скорость вращения (при разных значениях в setSpeed). В конечных точках поворота слышно тихое шебуршание двигателя. Питание внешнее 5V (одинаково работает и при питании от пина 5V и пина 3V3). При подаче питания иногда первый поворот в боковую позицию (крайняя по часовой стрелке - видимо инициализация объекта) быстрый, пауза 4-5 сек - потом работает цикл loop() - медленные повороты с паузой 2 сек.  Компилировалось как WEMOS D1 R2 & mini  и как NodeMCU 1.0 (с полной чисткой Flash: All Flash Contens).

Joner11, 2024-03-10:
Приветствую. Вот и я столкнулся с той же проблемой на esp8266 wemos. Кому то удалось осуществить поворот на 180 ?

aleta2020, 2024-03-10:
Победить esp8266 так и не удалось :-( . Пока перешел на Nano или ESP32 - на этих МК с ServoSmooth нет никаких проблем.

Joner11, 2024-03-11:
Да, стандартная Servo работает корректно.